ph1p/ikea-led-obegraensad #6: Pong

ingmar, 2022-12-23:
[image]

c64emulator, 2024-01-11:
Es gibt ja bereits viele Plugins, bei denen der ESP mit sich selber spielt. Das ist doch langweilig...

Anregung: 
Die ESP32 können doch auch Bluetooth (der S3 und der C3 allerdings nur BLE). Mit der Smartphone-App "Arduino ESP Bluetooth - Dabble" und der Bluetooth-Lib für den ESP32, beides von STEMpedia ([https://ai.thestempedia.com/docs/dabble-app/gamepad-module/](https://ai.thestempedia.com/docs/dabble-app/gamepad-module/)) könnten die Spiele interaktiv werden.

ph1p, 2025-01-11:
Saved it here: https://github.com/ph1p/ikea-led-obegraensad/discussions/125. If you want to open it again, feel free to do so