ph1p/ikea-led-obegraensad #9: REST API & Text on screen

mlemarrec, 2023-01-05:
Congratulations for this project.
2 ideas for improvements:
- Add a REST API to be able to control the screen via home automation
- Offer the possibility to send a text to scroll on the screen

ph1p, 2023-01-07:
I like the idea! Maybe an endpoint to which you can send the matrix and a few settings. You can do that now with websockets, but you have to establish a connection first.

mlemarrec, 2023-01-16:
Hello

Just a small contribution to separate the control part of the panel from the rest of the application and add compatibility with Adafr_it GFX (possibilité to print text): https://github.com/mlemarrec/ObegransadPanel
If you are interested I can try to integrate it into your project.

mlemarrec, 2023-01-16:
Just a small contribution to separate the control part of the panel from the rest of the application and add compatibility with Adafruit_GFX: https://github.com/mlemarrec/ObegransadPanel

d00mfish, 2023-08-01:
I was also trying to implement this but as my cpp skills are not the best I  still dont have it working.
I would loooove a REST API to send my current solarpanel panel power as a by time growing bar graph. The posibility to send a Json containing the array would be more than enough for me.