ph1p/ikea-led-obegraensad #14: Reconnect to WiFi in case of disconnection

craftbyte, 2023-01-11:
At the moment, if WiFi connection is lost, the ESP doesn't try to reconnect. This means the device has to be rebooted every time. Can something like the solutions at https://randomnerdtutorials.com/solved-reconnect-esp32-to-wifi/ be implemented?

ph1p, 2023-01-12:
Sure, I'll take a look at it

ph1p, 2023-07-05:
Closed, because of #41

closed 2023-07-05 by ph1p

c64emulator, 2024-04-12:
Das Problem scheint nicht behoben:
Leider musste ich bei meinem neuen WLAN-Router einen Scheduler einrichten, der in jeder Nacht (nachdem die FRITZ!Box einen DSL-Reconnect durchgeführt hat) den Router neu bootet. Das war notwendig, weil das WLAN extrem unzuverlässig lief. 
Als Folge davon öffnet das LED-Panel nun seinen AP und wartet auf Eingaben.