ph1p/ikea-led-obegraensad #36: Banner mode [suggestion from my kid!]

juliendorra, 2023-04-29:
One of my kids suggested that we could leave messages on the panel when I explained that the whole house has access to the control web page

As there's not enough resolution for a message board, a banner could be a way to pass messages. 

It would also be a cool way to use our panels to display classical conceptual art pieces like the ones from Jenny Holzer 😃 ("PROTECT ME FROM WHAT I WANT")

There's https://github.com/MajicDesigns/MD_Parola  but it is targeted as hardware controlling library

This much more basic project use 7 pixels by 5 characters (we should probably skip every other pixel vertically just like on the clock, considering the panel a 32x16 screen) https://projecthub.arduino.cc/OliB150/0988c46a-34ae-4d70-89a2-d38c4746bd6f

_(I'm posting a few issues as much for everyone here than for me, to keep a list of the enhancements I'd like to try and implement)_

kohlsalem, 2024-01-15:
@ph1p, I would arrgue, with the merge of https://github.com/ph1p/ikea-led-obegraensad/pull/69 this can be closed...

kohlsalem, 2024-01-15:
Even so i would say a "Banner Plugin" where you just manually give a text in the web interface could be nice