ph1p/ikea-led-obegraensad #38: tixy.land inspired mode [enhancement]

juliendorra, 2023-05-06:
A new creative coding mode inspired by tixy.land

**Context**
tixy (https://tixy.land/) is a very simple creative coding playground. Users can write a short formula using the time, index, x and y parameters, and the formula is evaluated and returned for each dot, and values are capped between -1 and 1. 0 to 1 means a white dot. 0 to -1 a red dot, with 1/-1 the max size and 0 no dot.

More infos there: https://twitter.com/aemkei/status/1323399877611708416

The code is simple enough and easy to ready:
https://github.com/aemkei/tixy/

**Limitation**
We of course drop the color (negative values) dots. 
Using persistence of vision and a fast refresh rate could help turning the dot size value into LED intensity values. 

**Adaptation**
We _could_ store and compute the formula on the micro-controller (using https://github.com/owenmcateer/tixy.land-display and https://github.com/codeplea/tinyexpr for example)

But much probably the easiest way to get a working version with reasonable effort is reuse and adapt the web front of tixy.land: 

 - use the computed value by the web page, 
 - send the values to the micro-controller as a stream
 - the micro-controller then drive the LEDs, acting as just a dumb receiver.
 
Plus: The preview and interpreter is already there, and the receiving code on the micro-controller can be simpler, focusing on driving the LEDs. All the existing tixy formulas shared on twitter and elsewhere can be tried.

Minus: There’s nor code to persist on the micro-controller, and thus once we close our browser the screen stops displaying our tixy formula results. We needs a browser to control and send the value, all the creative coding part is actually done in the browser, which may limit the hacking aspect / fun? The micro-controller needs to store and serve the tixy page and code.

juliendorra, 2023-05-11:
First version of tixy for obgraensad led
https://github.com/juliendorra/ikea-led-obegraensad/tree/tixy

https://github.com/ph1p/ikea-led-obegraensad/assets/109677/75cddb50-f83b-4453-b6c1-c1705f10ded7

Demo at refresh period of 3s (lower crash the ESP, see below)

https://github.com/ph1p/ikea-led-obegraensad/assets/109677/e31dbb82-2e60-42f7-a5c2-a6b05c958970

This branch replace the whole front end with a single tixy page. Only the js file is modified compared to the official tixy.land site.

The web page then use the seb socket to send screen events with the whole screen. 

- This version has a send period of 5000 ms. Lower period than eg. 3s freeze the ESP. Probably because of this unresolved bug in ESPAsyncWebServer: https://github.com/me-no-dev/ESPAsyncWebServer/issues/394

- no negative (red dots) values. Values are capped 0-1

- 0 to 1 tixy values are turned into 0 or 1 with a 0.5 cut off

At the moment the bottleneck is apparently the ESPAsyncWebServer server that cannot receive the data fast enough and crash.

ph1p, 2025-01-11:
Saved it here: https://github.com/ph1p/ikea-led-obegraensad/discussions/125. If you want to open it again, feel free to do so

closed 2025-01-11 by ph1p