ph1p/ikea-led-obegraensad #48: Any idea how to check if leds are working?

grzegorztomasiak, 2023-10-27:
I successfully wired all connections to the Obegraensad and was working fine for 2 days. I had set clock to display.
And today I saw that LED hung on minutes (hours were already gone, not lit). So I restarted ESP32 and software is running fine under my ip address but no leds are lit.

I tried to program esp32 again but still no luck with LEDs. I suspect something burned inside. Anyone has idea how to lit up all leds on the led boards?

grzegorztomasiak, 2023-11-11:
noone?

cixio, 2023-11-11:
> https://jmp.sh/s/3fFVJ6XEmGT6IXQAnNdF
File not found

It is challenging to assess whether something is defective or not functioning correctly from a distance. Check the connections and settings, especially the GPIO. If possible, please provide photos of the connections and screenshots of your settings.

The LEDs need to be controlled, and there is no way to test them without proper control signals. If you suspect that a board is defective, disconnect the cables connected to it and try the next board. While the LEDs may not behave entirely correctly, there should be some illumination at the very least.

schw4rzlicht, 2023-11-13:
I try and add some "init" code that blinks all LEDs on startup. When I was hacking my lamp I would have needed something like this as well since my GPIO wasn't working properly but I was unsure whether it was software or hardware related. I might come back with a PR soon.

schw4rzlicht, 2023-11-13:
Et voila.

djboris9, 2023-11-15:
I'm not using this project for my setup but my experience was that when Vcc is +5V and the ESP32 GPIOs have a potential of +3.3V, it's just on the edge to communicate to the SCT2024.

I also had a setup running for some time, which then showed random LED. Traced with a logic analyzer I could verify that the SCTs weren't working as expected. Having the communication lines at slightly below Vcc, everything works as expected.

So you could try to manually communicate to the device on a higher level than 3.3V.

grzegorztomasiak, 2023-11-15:
https://gifyu.com/image/SRsyt
I've updated gif showing how the leds are behaving after connecting power supply. ESP is not wired at the moment, so it's not a problem with ESP itself but rather something else. After connecting power supply with 5V (mobile charger), leds flashes for split of a second and nothing else. Wiring ESP does not help either because behavior is the same.

As I mentioned in the first post, everything was working fine and hanging on my wall for 2 days and suddenly stopped working.

dfleckenstein, 2024-03-23:
@grzegorztomasiak I have the same problem. Obegraensad was working fine for about a month and suddenly stopped working. After connecting USB cable, some leds flash for split of a second and they are gone. Unfortunately, I don't have found a solution.

ph1p, 2025-01-11:
Saved it here: https://github.com/ph1p/ikea-led-obegraensad/discussions/125. If you want to open it again, feel free to do so

closed 2025-01-11 by ph1p