ph1p/ikea-led-obegraensad #49: crash on NodeMCU V2 Illegal instruction Screen_::getRotatedRenderBuffer() at src/screen.cpp:25

andreasziegl, 2023-11-09:
Hi, 
I get random crashes when I am using Star. Does someone know where this could come from? Weather also fails, but with a different Exception. 

Maybe @jekkos as you have added  NodeMCU V2 support?

Board: NodeMCU Amica Modul V2 ESP8266 ESP-12F
Pio version: PlatformIO Core, version 6.1.11
OS: Windows 10

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x40207284 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Illegal instruction
  epc1=0x40207284 in Screen_::getRotatedRenderBuffer() at src/screen.cpp:25        

>>>stack>>>

ctx: sys
sp: 3fffec10 end: 3fffffb0 offset: 0150
3fffed60:  4010078c 40100740 00000020 40100573  
3fffed80:  40101678 00000008 3fffc230 4000050c  
3fffed90:  40004858 00000030 00000020 ffffffff  
3fffeda0:  40101fc0 00e8a101 3feffe00 60000200
3fffedb0:  fffdffff 04000001 fffffffe 04000001
3fffedc0:  04000003 00000001 04000002 00000008
3fffedd0:  00000000 00000000 60000600 00000030
3fffede0:  4023e640 00000001 00000000 40100960
3fffedf0:  3fffc200 3fff0649 3fff0548 4010054f
3fffee00:  646e6173 40100740 00000020 40100573
3fffee10:  40252b62 00000001 00000000 40100960
3fffee20:  3fffc200 3fff0649 3fff0548 4010054f
3fffee30:  00000000 00000000 0000001f 40100624
3fffee40:  40252202 00000020 3fffc228 401062a9  
3fffee50:  4000050c 7fffffff 3fffc230 4000050c
3fffee60:  40000f68 00000030 00000010 ffffffff
3fffee70:  40000f58 00000000 00000020 00000000
3fffee80:  00000002 4024f788 3ffefc28 00000001
3fffee90:  4024c461 3fff0cc0 00000000 3fffdab0
3fffeea0:  00000000 3fffdcb0 3ffefc78 00000030
3fffeeb0:  00000000 00000020 3fff1584 40247800
3fffeec0:  00000008 00000000 f7ffffff 60000600
3fffeed0:  00000008 00000000 f7ffffff 4023192d
3fffeee0:  4024f2c4 00000008 3ffefc50 06efd0dd
3fffeef0:  4024f6d6 73f7ced9 0001c699 60000600
3fffef00:  06efc76a 3ffefc50 4024f79b 3ffea7dc
3fffef10:  3ffefc50 022ac5e7 3ffefc50 60000600
3fffef20:  402569b9 3ffea7dc 3ffefc50 06efc76a
3fffef30:  402569fe 3fffdab0 00000000 3fffdcb0
3fffef40:  3ffefc60 40000f65 3fffdab0 00000000
3fffef50:  40000f49 3ffeab10 ffffff00 40000f49
3fffef60:  40000e19 00078ee8 bff00000 0000bfff
3fffef70:  00000005 aa55aa55 00000088 40105b11
3fffef80:  40105b17 bff00000 0000bfff 401011ac
3fffef90:  4010000d 74616577 43726568 401000ab
3fffefa0:  40246cb8 3fffef3c 40246c69 3ffffdf8
3fffefb0:  3fffffc0 00000000 00000000 feefeffe
3fffefc0:  feefeffe feefeffe feefeffe feefeffe
3fffefd0:  feefeffe feefeffe feefeffe feefeffe
3fffefe0:  feefeffe feefeffe feefeffe feefeffe
3fffeff0:  feefeffe feefeffe feefeffe feefeffe
3ffff000:  feefeffe feefeffe feefeffe feefeffe
3ffff010:  feefeffe feefeffe feefeffe feefeffe
3ffff020:  feefeffe feefeffe feefeffe feefeffe
3ffff030:  feefeffe feefeffe feefeffe feefeffe
3ffff040:  feefeffe feefeffe feefeffe feefeffe
3ffff050:  feefeffe feefeffe feefeffe feefeffe
3ffff060:  feefeffe feefeffe feefeffe feefeffe
3ffff070:  feefeffe feefeffe feefeffe feefeffe
[... 298 lines cut ...]
0x40100740 in timer1_isr_handler at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_timer.cpp:34
0x4010078c in timer1_isr_handler at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_timer.cpp:44
0x4021bef8 in esp_suspend at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:130
0x4021bfdf in __esp_delay at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:155
0x4021bfd9 in __esp_delay at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:155
0x4021a258 in random(long) at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WMath.cpp:48 (discriminator 4)
0x401059f7 in lmacTxFrame at ??:?
0x401063ed in ets_timer_arm_new at ??:?
0x4021beec in esp_get_cycle_count at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:64
 (inlined by) EspClass::getCycleCount() at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Esp.h:221
 (inlined by) esp_suspend_within_cont at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:122
 (inlined by) __esp_suspend at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:128
0x4021bff6 in __esp_delay at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:161
0x4021c229 in esp_try_delay(unsigned int, unsigned int, unsigned int) at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:182 (discriminator 4)
0x4010054f in Screen_::_render() at src/screen.cpp:165
0x40100960 in digitalWrite at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring_digital.cpp:87
0x4021c790 in esp_delay<__delay(long unsigned int)::<lambda()> > at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/coredecls.h:69
 (inlined by) esp_delay<__delay(long unsigned int)::<lambda()> > at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/coredecls.h:78
 (inlined by) __delay at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:39
0x40203e12 in stars() at src\mode/stars.cpp:21
0x402026fc in loopOfAllModes() at src\mode/mode.cpp:247
0x4020116a in loop at src/main.cpp:77
0x4021c07a in loop_wrapper() at C:\Users\Andi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:258

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Menschomat, 2023-11-10:
Same here. Not only for stars it was extremly instable on NodeMCU. 
My easy "solution" was to switch to ESP32.

jekkos, 2023-11-10:
could you try to remove the delay at stars:21?

the issue seems to be related to isr1 timer.

jekkos, 2023-11-10:
If removing the delay does not help, you can try to increase the interrupt timer to 1000 eg to see if it runs more stable.

I read that if interrupts run too long it might destabilize the chip as it also runs wifi functions in async.

the ESP8266 was much more stable without the per pixel brightness

jekkos, 2023-11-10:
If that does not work I can push my old version that has storage and global brightness enabled but no support for per pixel brightness.

andreasziegl, 2023-11-10:
@jekkos thank you for checking! I will try it later today or over the weekend

jekkos, 2023-11-13:
Hi today someone posted an issue on a possible bug, which led to check the problem here a bit further. The fact that you get an illegal instruction might because of a memory corruption issue due to integer overflow. 

You could try to disabe the pixel brightness by removing the multiplcation here https://github.com/ph1p/ikea-led-obegraensad/blob/5ad09fa4c06fedfa00d01e3898202d4ef1ea0244/src/screen.cpp#L19 and see if that solves the crash

jekkos, 2023-11-25:
@andreasziegl any idea if the issue is fixed with latest master?

ph1p, 2025-01-13:
I will close this. Please reopen if its no fixed

closed 2025-01-13 by ph1p