ph1p/ikea-led-obegraensad #58: Noob need some help.

s0m3thin, 2023-11-17:
Hey! Thanks for this project. It made me to buy a lamp like this and an ESP32 and get started.
The ESP is connected to my PC right now. I also got VScode and i downloaded platformIO.

But could you please get more in detail by adding the vars to the .h files and get the firmware onto the device? or a hint how to? that would be great.

ph1p, 2023-11-17:
Hello @s0m3thin 

Have a look at the "How to" section. If you are missing something in description, feel free to open a PR.

**I'll try to describe the steps in a bit more detail:**

- First you need to find out which board you have and possibly change the `platform.ini` file. In this file you can define your board and the libraries, because every board is different and might need different compatible libraries
- Solder the pins of your board to the lamp
- Connect the board to your computer
- run `git clone https://github.com/ph1p/ikea-led-obegraensad.git` in your terminal or download the file as a zip and unzip it
- open the folder in VSCode
- Create a `secret.sh`
 file in the `include` folder and copy the code from the "how to" section
- click on this symbol in VSCode
[image]
- done :)

BrookeDot, 2023-11-18:
I was new to Platform.io (but not Arduino) and found that the [Quick Start](https://docs.platformio.org/en/latest/core/quickstart.html) was quite helpful. If you have no idea what board to select, maybe post a picture here and I or someone else can help out.

m0uh85, 2023-11-18:
Hello, I need also help.
I have a D1 Mini (ESP8266) and a D1 Mini (ESP32).
Im compiled both Versions from the Script.
The ESP8266 is connected to the WiFi and "works" but after a while the Controller stucks.
So tried the same Code with the ESP32 but its not working no Web frontend.
I tried to modifie the Code and put in the envd for the D1 Mini but same Issue. I tried 3 different Controller Just to be Sure. 
Can someone help me to Analyse were the Fault is?

BrookeDot, 2023-11-18:
> So tried the same Code with the ESP32 but its not working no Web frontend.

Is the device connecting to the WiFi network? What does "not working" mean in this case? Can you flash a simple test pattern like[ in this PR](https://github.com/ph1p/ikea-led-obegraensad/pull/53)?

m0uh85, 2023-11-18:
No it is not connected to the wifi (but wifi setting are ok because it works on the esp8266)
With the serielle Monitor it looks like a bootloop or something.

Blubbaa, 2023-11-19:
Can confirm, on `esp32-c3-devkitc-02` board the program is stuck in a boot loop:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x918
load:0x403ce710,len:0x25f4
entry 0x403cc710
```
Any ideas how to fix this?

m0uh85, 2023-11-20:
> Can confirm, on `esp32-c3-devkitc-02` board the program is stuck in a boot loop:
>
> ```
> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
> Saved [...]

Ohh no I ordered this board because I thought maybe my D1 Mini Board is the issue :(

ph1p, 2023-11-20:
> Can confirm, on `esp32-c3-devkitc-02` board the program is stuck in a boot loop:
>
> ```
>
> ESP-ROM:esp32c3-api1-20210207
>
> Build:Feb  7 2021
>
> rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT) [...]

Try to change the default pins, maybe this will help

m0uh85, 2023-11-20:
Before I downloaded the zip manually.
Now I tested to clone it with git like you explained before. I deleted the old folder and everything.... now it looks like that its working for me!

hendrikreuhl, 2023-11-25:
I'm having the same problem that I'm stuck in the bootloop with my seeed studio xiao esp32c3.

Any ideas on how to fix this?

Blubbaa, 2023-11-26:
So its definitely the PIN setup. Changing the default ones to some free GPIOs of the board did the trick. See also https://github.com/ph1p/ikea-led-obegraensad/issues/60 for Lolin D32 Pro.

closed 2025-01-10 by ph1p