ph1p/ikea-led-obegraensad #60: Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled + Solution

MoJo2600, 2023-11-18:
My ESP was crash looping after i flashed the program with platform.io on my `Lolin D32 Pro`. I was tearing my hair out, but i found the culprit. 

The exception looks something like this:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008f908  PS      : 0x00060b30  A0      : 0x800e98d9  A1      : 0x3ffcbba0  
A2      : 0xffffffff  A3      : 0xffffffff  A4      : 0xffffffff  A5      : 0x00060b23  
A6      : 0x00060b23  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000004  
A10     : 0x00060b23  A11     : 0x3ffc41f4  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x007bf5a8  A15     : 0x003fffff  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003f  LBEG    : 0x4008b961  LEND    : 0x4008b971  LCOUNT  : 0xfffffff4  

Backtrace: 0x4008f905:0x3ffcbba0 0x400e98d6:0x3ffcbbe0 0x400ea165:0x3ffcbc30 0x400e5c98:0x3ffcbc60 0x400e3005:0x3ffcbc80 0x400e35ab:0x3ffcbca0 0x400e35bd:0x3ffcbcc0 0x400d3057:0x3ffcbce0 0x400d35d0:0x3ffcbd10 0x400e628e:0x3ffcbd40
```

I figured out, that `GPIO16` that is used for the button is also used for the serial `RX`. This causes the firmware to crash every time something is written to the console. The code is working when I either change the button pin or If i remove all serial outputs.

[image]

My recommendation would be to change the default pin for the button to something else.

ph1p, 2023-11-19:
Nice find :) Maybe you could just expand the readme. There is already a table for different boards and their pins

MoJo2600, 2023-11-19:
Good point, I'll have a look

MoJo2600, 2023-11-24:
I updated the Readme with a pinout. Have a look here: https://github.com/ph1p/ikea-led-obegraensad/pull/68

Maybe it would be a good idea to enable the wiki feature for this repository. Then there could be a page with details for the devices?

PR #68 (2023-11-28): chore: Update Readme for Lolin D32
Added PIN outs for D32 where using PIN 16 leads to a crash loop.

fixes #60
closed 2023-11-28 by ph1p