ph1p/ikea-led-obegraensad #63: Is somewhere a documentation of the websocket feature available?

KrasserMann, 2023-11-19:
Hi, and thank you for this gereat project. I just converted my obegraensad from stupid to useable ;)

I want to control this device by the websocket feature to switch the plugins or display some custom images. 

I tried to analyse the code of websocket.cpp but its not clear for me in what format the json data has to be sent to the websockt.

My smarthome ist based on SYMCON and uses mainly php code for scripting. The system is allready receiving Data from the ESP in this format:

19.11.2023, 18:58:21 |             RECEIVED | {"status":0,"plugin":8,"event":"info","rotation":0,"brightness":255}
19.11.2023, 18:58:05 |             RECEIVED | {"data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,0,0,255,255,255,0,0,0,0,0,0,0,0,0,255,255,0,0,255,0,255,0,0,0,0,0,0,0,0,0,0,255,0,0,255,255,255,0,0,0,0,0,0,0,0,0,0,255,0,0,255,0,255,0,0,0,0,0,0,0,0,0,0,255,0,0,255,255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,255,255,255,0,0,255,255,255,0,0,0,0,0,0,0,0,255,0,0,0,0,255,0,255,0,0,0,0,0,0,0,0,255,255,255,0,0,255,255,255,0,0,0,0,0,0,0,0,0,0,255,0,0,255,0,255,0,0,0,0,0,0,0,0,255,255,255,0,0,255,255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"status":0,"plugin":11,"event":"info","rotation":0,"brightness":255,"plugins":[{"id":1,"name":"Draw"},{"id":2,"name":"Breakout"},{"id":3,"name":"Snake"},{"id":4,"name":"GameOfLife"},{"id":5,"name":"Stars"},{"id":6,"name":"Lines"},{"id":7,"name":"Circle"},{"id":8,"name":"Rain"},{"id":9,"name":"Firework"},{"id":10,"name":"Big Clock"},{"id":11,"name":"Clock"},{"id":12,"name":"Weather"},{"id":13,"name":"Animation"}]}

Maybe you can give me some support to this!

Danke!! Gruß Martin

KrasserMann, 2023-11-19:
Small Update,  i managed to switch the plugins by setting "event" => "plugin" and use the desired plugin number.

Setting the brightness works to. but i was not able to send a custom image by sending the full RenderBuffer. 

I think im using the wrong JSON string to get this working

Martin

ph1p, 2023-11-20:
The easiest way is to send it as a binary https://github.com/ph1p/ikea-led-obegraensad/blob/2abb613edad4168b3f4c4b2384b93d37d7b7238e/src/websocket.cpp#L72 . You can also create a new plugin with a websocketHook and define some custom websocket payloads https://github.com/ph1p/ikea-led-obegraensad/blob/2abb613edad4168b3f4c4b2384b93d37d7b7238e/src/plugins/DrawPlugin.cpp#L24

KrasserMann, 2023-11-22:
Ok, thank you,  i spent some time in trying to send the binary data via websocket to the esp but allways there ist no response on my data.

The ESP does not throw an JSON dserialisation error, so i think my data arrives as binary but nothing happens.

In the websocket Code there are three requirements for accepting the binary data.

info->opcode == WS_BINARY && currentStatus == WSBINARY && info->len == 256

The Last one is the number of bytes i sent, but what is WS_BINARY? Where does this come from? 

I think i neet to add some json event in which i can submit the Screenbuffer Data as a String like "100001001010010100...." 

Martin

Janideluxe, 2023-11-22:
> Hi, and thank you for this gereat project. I just converted my obegraensad from stupid to useable ;)
>
> I want to control this device by the websocket feature to switch the plugins or display some [...]

how do you switch the plugins via websocket? I'm running an Iobroker and want to change the Plugins with it, but dont know what i have to send and to which adress.

KrasserMann, 2023-11-22:
Hi Janideluxe, i connected my websocket client to ws://ip-address/ws.

To switch the plugin you need to send the following text data via the websocket connection:

{"status":0,"plugin":1,"event":"plugin","rotation":0,"brightness":255}')

Important is the event key, to change the plugin it has to be plugin, and then you can send the plugin number in the plugin key,

Martin

KrasserMann, 2023-11-22:
Now i worked arround my binary problem by adding the following code to the websocket.cpp file after line 120 so i can post an pattern of 256x  010101 in the bin key by using bindata as event key.

`else if (!strcmp(event, "bindata"))
          {
            
           const char *datastring = wsRequest["bin"];
           int len = strlen(datastring);
           
           Serial.print("Data Bytes: ");
           Serial.println(len);
           if ( len != 256) {
            Serial.print("Falsche Anzahl Data Bytes: ");
            Serial.println(len);

           } else { 
            for (int i=0;i<len;i++) {
              
              
              if (datastring[i] == '1' ){
                data[i]=255;
              } else {
                data[i]=0;
              }
            }
            Serial.println("Displaying Bin Pattern ");
            Screen.setRenderBuffer(data, true);
            data[len]=0;
           }

           

          } `

Martin