ph1p/ikea-led-obegraensad #67: Feature: Auto-Dim in the evening

BrookeDot, 2023-11-22:
One of my favorite features of my [Tidbyt](https://tidbyt.com) is that is has features to:
- Dim in the evening
- Display a dim clock (opposed to cycling apps) in the evening. 

I would like to enable such a feature for my Obegraensad. I've been thinking about the best way to implement this for a while and am not thinking a "Plugin" make sense as I'd want to set the brightness in the loop. 

Currently I am thinking of adding a few new constants for `MIN_BRIGHTNESS`, `DIM_TIME`, `ON_TIME`. I also thought of getting the sunset time based on the GPS coordinates which would also be fairly straightforward but feels a bit "over engineered" to me as setting a fixed time allows more control. Maybe that's a V2 option to allow `SUNSET` to be defined within the auto-dimming setting.

Things do get slightly more complicated with the web app component of this allowing brightness to be set. I'm wondering it it makes sense to offer a checkbox for "auto dim" in the web app which disables the brightness or if we can just recalculate the "steps" to `MIN_BRIGHTNESS` when the brightness level changes. 

I mean as I type this, I was originally thinking the screen would slowly dim over time. However, the simplest approach would just be to dim to a certain level at a given time. 

In short, what I am saying/asking is if this feature would be useful for others and if so if they have thoughts on the best approach before I start to tinker. 

Related: #66

jekkos, 2023-11-23:
I installed home assistant two weekends ago and I'm quite fond of it now. Most of the triggers you describe are already implemented there. If the matrix can be exposed as an entity then a platform like that can make automations like this in a couple of clicks. To me that seems less work while it would also create other possibilities.

But I agree that dimming it at night would be a nice feature.

grzegorztomasiak, 2023-11-24:
> One of my favorite features of my [Tidbyt](https://tidbyt.com) is that is has features to:
 Wow I did not know about Tidbyt, Im gonna order few! thanks man.

jekkos, 2023-11-27:
> Wow I did not know about Tidbyt, Im gonna order few! thanks man.

I have now found this one below
https://blakadder.com/ulanzi-pixel-clock/

its a lot cheaper and esp32 already installed. You can run pixelit software on this it makes it very nice to use

kohlsalem, 2023-12-11:
Hmm, i could have added here instead of making a new issue: https://github.com/ph1p/ikea-led-obegraensad/issues/74

I think a light sensor is the best way to handle the dimming.

MauiKano, 2023-12-14:
I did add a PIR sensor. If a person is in its range, the LED panel is switched to maximum brightness. After a configurable time a timer event triggers a ISR that reduces the brightness to a minimum (or could completely dim). To me, that works even better than just dimming in the evening. You find my mods on my [fork of the code ](https://github.com/MauiKano/ikea-led-obegraensad).

ph1p, 2025-01-11:
Saved it here: https://github.com/ph1p/ikea-led-obegraensad/discussions/125. If you want to open it again, feel free to do so