ph1p/ikea-led-obegraensad #70: Flickering Panel with ESP32C3

hendrikreuhl, 2023-11-26:
Hey everyone,

first, thanks for the amazing development on this project.

I'm facing issues when using the Seeed-Xiao-ESP32C3

The pins i'm using:
```
PIN_ENABLE 3 //GPIO 3
PIN_DATA 20 // GPIO 20
PIN_CLOCK 7 // GPIO 7
PIN_LATCH 5 // GPIO 5
PIN_BUTTON 6 // GPIO 6
```

The software starts an the webinterface is also reachable.
When setting a mode (clock e.g.) some leds light up at the correct positions.
Some pixels are missing and do only light up if you dim the screen multiple times. Sometimes the leds also flicker.
Auto refreshing (on a new minute e.g.) does not work, random pixels light up sometimes.

When taking a look on the serial monitor, I get these error messages sometimes:

`[  2258][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected`
and
`[  2263][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: data NOT_FOUND`

Does someone have any idea on how to fix this or might this be a board related issue?

Thanks

wendehals, 2023-12-06:
Hello,

I had the same problem of flickering pixels. I solved the issue by completely removing the old controller chip. Beforehand I only desoldered the VCC pin of the old controller chip.

Greetings

dsancheznet, 2023-12-08:
Hi guys,
I had the same problem and after almost driving me crazy I seitched the power source and now the problem is completely gone.
My guess is this appears with poor soldering contacts or a bad power brick.

Best regards
D.Sanchez

gelse, 2024-01-28:
> My guess is this appears with poor soldering contacts or a bad power brick.

I had the same problem and solved it by re-soldering the contacts.
So - i would propose this issue to be closed, as it is NOT a software-bug.

BrookeDot, 2024-01-28:
Perhaps the solution to mark this as "closed" is clarify in the readme file the suggested power supply and if flickering happens to check the contact? That seems like an easy win to help more people use this project and solve the problem themselves.

ph1p, 2024-01-29:
Thanks for your suggestions. I added a "Trouble shooting" section to the readme file :)

commit c6397699 (2024-01-29): chore(docs): add trouble shooting section

Remove the "Ideas" section and deal with them under "Issues" or "Discussion
closes #70
closed 2024-01-29 by ph1p