ph1p/ikea-led-obegraensad #73: Panel Freezes when too many pixels are lit?

YoshiFan501, 2023-12-04:
I was trying to make a simple plugin that turns on every LED to make it a simpe lamp.

When I wrote my code for that, the led panel flashed on and off, frozen until i reset the board.

My code was a simple for loop in the setup that lit every pixel, and a delay in the main loop.

I wanted to test with the draw plugin, and after I got here, my esp froze and dropped wifi connection

[image]

Is it a board issue, or is something wrong in the code? I do not believe I have had this issue before with earlier firmwares. A simple all LEDs on lamp has been on my want list for a while, and was hoping to make it a reality.

YoshiFan501, 2023-12-04:
`#include "plugins/LampPlugin.h"

LampPlugin::LampPlugin() {
    // Constructor logic, if needed
}

LampPlugin::~LampPlugin() {
    // Constructor logic, if needed
}

void LampPlugin::setup() {
  for (int row = 2; row < (ROWS - 2); row++)
  {
    for (int col = 2; col < 14; col++)
    {
      Screen.setPixel(col, row, 1);
    }
  }
}

void LampPlugin::loop() 
{
  delay(100);
}

const char* LampPlugin::getName() const {
    return "Lamp"; // name in GUI
}`

BrookeDot, 2023-12-04:
What board are you using? It may be running out of memory (RAM).

BrookeDot, 2023-12-10:
Any logs?

YoshiFan501, 2023-12-10:
How would I get those?

ra-ma, 2023-12-12:
As a hint: Check the power supply. If more LED are lit, the power consumption ist considerable higher. I was measuring Something between 0,2 and over 1 amps depending on the number and brightness. 

Eventually then the voltage breaks down and the ESP gets stuck as it's not getting sufficient power anymore.

BrookeDot, 2023-12-12:
>  Check the power supply... I was measuring Something between 0,2 and over 1 amps depending on the number and brightness.

That's a good point. The IKEA install guide says:
> Used with Class 2 listed power supply from IKEA with output 5Vdc, min. 1A

I am using a 1A power supply I bought at the same time as my OBEGRÄNSAD. So worth trying for sure.

YoshiFan501, 2023-12-28:
I tried a 2.5A USB and it seems to work now

closed 2023-12-28 by YoshiFan501