ph1p/ikea-led-obegraensad #78: weather screen - watchdog timeout issue

robotfishe, 2023-12-09:
Just updated my Obegransad controller to the latest version and I'm having a lot of crashes. It's usually when entering the weather plugin (bit embarassing for me as the original contributor of the weather screen lol). Seems like it has something to do with the async web server though, rather than the weather plugin itself. Here's the serial monitor output:

```
E (75333) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (75333) task_wdt:  - async_tcp (CPU 1)
E (75333) task_wdt: Tasks currently running:
E (75333) task_wdt: CPU 0: IDLE
E (75333) task_wdt: CPU 1: IDLE
E (75333) task_wdt: Aborting.

abort() was called at PC 0x400f34d5 on core 0

Backtrace: 0x40083925:0x3ffbeccc |<-CORRUPTED

  #0  0x40083925:0x3ffbeccc in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408

ELF file SHA256: 066afc2422b3e5a8

Rebooting...
```

Happening on two ESP32 boards. Anyone seen anything like this before?

kohlsalem, 2023-12-11:
I did not check this particular issue, but i noticed while displaying messages: if you block loop for more then a couple of seconds (5? 10? around that) you get forcefully aborted. Could this be the case?

robotfishe, 2023-12-11:
Thanks, that does help. I'm doing a bit of a rewrite of that plugin anyway, to add options for forecast values (inspired by me having the display next to my front door, where a current weather display is a bit pointless since I can just look outside, but knowing what the weather will be in an hour is very helpful). I'm thinking in the process I might rewrite the weather loop into a tick-tock system, where the loop triggers a download of new weather data from the API on one pass, then deserialises and analyses it on the next. Hopefully that'll stop the main loop from getting interrupted for too long.

closed 2023-12-11 by robotfishe