ph1p/ikea-led-obegraensad #79: Alternative button wiring

RBEGamer, 2023-12-12:
After unsoldering the old microcontroller on the PCB, it is possible to use the 7th pin for the button connection.
Using this the original wiring of the button can be the same. The other end of the button pin is already connected to ground on the PCB.

[image]

OldDroid, 2024-01-21:
Can confirm it works like a charm! Ty @RBEGamer

Elump, 2024-02-04:
An alternative is to use the left pad (in printing direction the lower pad) of C6. Soldering is easier here compared to uC pin.

RBEGamer, 2024-02-04:
[image]

Works, you can see the traces between the two pads.

closed 2024-04-05 by ph1p