ph1p/ikea-led-obegraensad #82: OTA not working

jekkos, 2023-12-20:
I have tried to use OTA updates a couple of times, but they never succeed. The upload does start but always stalls at some point. Then after that nothing happens and the task just stops. Does anyone have similar experiences? 
It would be nice to have this as otherwise I have to detach the lamp from the wall every time there is a new feature or bugfix.

thunder-62, 2023-12-22:
actually it works quite well here. The only thing I did was to fix an issue related to the weird Path on my PC:

    env.Execute("\"$PYTHONEXE\" -m pip install requests_toolbelt")
    env.Execute("\"$PYTHONEXE\" -m pip install tqdm")

I just put $PYTHONEXE in masked Quotes (\")  maybe this helps ;-)

Aeonos, 2023-12-25:
Can someone explain to me, how one can use the ota? I just managed to upload to my esp32 with the usb serial connection.

thunder-62, 2023-12-27:
just have look @ upload.py:
# Allows PlatformIO to upload directly to AsyncElegantOTA
#
# To use:
# - copy this script into the same folder as your platformio.ini
# - set the following for your project in platformio.ini:
#
# extra_scripts = platformio_upload.py
# upload_protocol = custom
# upload_url = <your upload URL>
#
# An example of an upload URL:

# upload_URL = http://192.168.1.123/update

the only remaining task left is to ensure unsername and password in upload.py matches  OTA_USERNAME and OTA_PASSWORD in include/secrets.h

fr3deric, 2024-11-16:
I know this issue is almost a year old, but I would like to add my 2 cents as I had the same problem as well lately. 

If the OTA fails, the firmware that needs to be installed is in your project directory under `.pio\build\esp32dev\firmware.bin`

You can then go to `http://<your-server-adress>/update` you can upload the firmware manually without having to detach the lamp. 

That's how I it worked for me. Hope it helps.

jekkos, 2024-11-16:
Thanks for the tip, was not aware of this. I managed to reach the interface and have uploaded a firmware file there. The device does not seem to reboot automatically? So did a hard reset and now I assume it's running on my updated version. Would be great if this works.

jekkos, 2024-11-24:
Seems to work this makes using the device very convenient.

closed 2024-11-24 by jekkos