ph1p/ikea-led-obegraensad #91: can not build - undefined reference to `HTTPClient'

marcogracklauer, 2024-01-20:
Hi,
i've pulled the current version, switched to "Default" -> "General" -> "build all" and get an error:

```
/Users/Marco/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/lib2c7/libHTTPClient.a(HTTPClient.cpp.o):(.literal._ZN10HTTPClient5beginE6String+0x8): undefined reference to `HTTPClient::begin(String, char const*)'
/Users/Marco/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/lib2c7/libHTTPClient.a(HTTPClient.cpp.o): in function `HTTPClient::begin(String)':
/Users/Marco/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:182: undefined reference to `HTTPClient::begin(String, char const*)'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
================================================================= [FAILED] Took 20.61 seconds =================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:20.610
============================================================ 1 failed, 0 succeeded in 00:00:20.610 ============================================================
```

When I switch to an other board, there are other errors, for example:

```
Compiling .pio/build/nodemcuv2/src/plugins/WeatherPlugin.cpp.o
src/plugins/BigClockPlugin.cpp: In member function 'virtual void BigClockPlugin::loop()':
src/plugins/BigClockPlugin.cpp:19:29: error: 'getLocalTime' was not declared in this scope
   if (getLocalTime(&timeinfo))
                             ^
src/plugins/ClockPlugin.cpp: In member function 'virtual void ClockPlugin::loop()':
src/plugins/ClockPlugin.cpp:19:29: error: 'getLocalTime' was not declared in this scope
   if (getLocalTime(&timeinfo))
                             ^
src/messages.cpp: In member function 'void Messages_::scrollMessageEveryMinute()':
src/messages.cpp:73:31: error: 'getLocalTime' was not declared in this scope
     if (getLocalTime(&timeinfo))
                               ^
src/messages.cpp: In function 'void handleMessage(AsyncWebServerRequest*)':
src/messages.cpp:136:25: error: 'stoi' is not a member of 'std'
         graph.push_back(std::stoi(token));
                         ^
src/plugins/PongClockPlugin.cpp: In member function 'void PongClockPlugin::drawCharacter(int, int, std::vector<int>, int, uint8_t)':
src/plugins/PongClockPlugin.cpp:12:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < bits.size(); i += bitCount)
                                 ^
src/plugins/TickingClockPlugin.cpp: In member function 'virtual void TickingClockPlugin::loop()':
src/plugins/TickingClockPlugin.cpp:11:29: error: 'getLocalTime' was not declared in this scope
   if (getLocalTime(&timeinfo))
                             ^
src/plugins/PongClockPlugin.cpp: In member function 'virtual void PongClockPlugin::setup()':
src/plugins/PongClockPlugin.cpp:197:29: error: 'getLocalTime' was not declared in this scope
   if (getLocalTime(&timeinfo))
                             ^
src/plugins/PongClockPlugin.cpp: In member function 'virtual void PongClockPlugin::loop()':
src/plugins/PongClockPlugin.cpp:216:33: error: 'getLocalTime' was not declared in this scope
       if (getLocalTime(&timeinfo))
                                 ^
*** [.pio/build/nodemcuv2/src/plugins/BigClockPlugin.cpp.o] Error 1
*** [.pio/build/nodemcuv2/src/plugins/ClockPlugin.cpp.o] Error 1
src/main.cpp: In function 'void connectToWiFi()':
src/main.cpp:117:8: error: 'class ESP8266WiFiClass' has no member named 'setHostname'
   WiFi.setHostname(WIFI_HOSTNAME);
        ^
*** [.pio/build/nodemcuv2/src/messages.cpp.o] Error 1
*** [.pio/build/nodemcuv2/src/plugins/PongClockPlugin.cpp.o] Error 1
*** [.pio/build/nodemcuv2/src/plugins/TickingClockPlugin.cpp.o] Error 1
*** [.pio/build/nodemcuv2/src/main.cpp.o] Error 1
========================================================= [FAILED] Took 2.40 seconds =========================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:02.401
==================================================== 1 failed, 0 succeeded in 00:00:02.401 ====================================================
```

I've tried on a fresh pull and one with some modifications as mentioned in the instructions but still similar errors. 
What's wrong?

Janideluxe, 2024-01-21:
`src/messages.cpp: In function 'void handleMessage(AsyncWebServerRequest*)':
src/messages.cpp:136:25: error: 'stoi' is not a member of 'std'
         graph.push_back(std::stoi(token));`

to eliminate this error for me worked to remove "stoi" and add "atoi"

ph1p, 2025-01-13:
I will close this for now. Feel free to open it again, if you need any help

closed 2025-01-13 by ph1p

marcogracklauer, 2025-01-14:
Im still not able to build the project on my MacBook Pro M1 in Visual Studio. Copied the cloned directory to a Windows 11 VM and the build is successful. 

```
compiling .pio/build/esp32dev/lib708/ESP Async WebServer/WebServer.cpp.o
src/webhandler.cpp: In function 'void handleGetData(AsyncWebServerRequest*)':
src/webhandler.cpp:114:34: error: exception handling disabled, use -fexceptions to enable
     catch (const std::exception &e)
                                  ^
src/webhandler.cpp:116:42: error: 'e' was not declared in this scope
         request->send(500, "text/plain", e.what());
                                          ^
Compiling .pio/build/esp32dev/lib0e4/DNSServer/DNSServer.cpp.o
Compiling .pio/build/esp32dev/lib9d9/ESPmDNS/ESPmDNS.cpp.o
*** [.pio/build/esp32dev/src/webhandler.cpp.o] Error 1
========================================================== [FAILED] Took 2.54 seconds ==========================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:02.543
==================================================== 1 failed, 0 succeeded in 00:00:02.543 ====================================================
```
As i have the Windows VM, i can use the project. But debugging on the MacBook was a little frustrating until I've tried the Windows Version and it works out of the box.