ph1p/ikea-led-obegraensad #92: Bootloop ESP32

Janideluxe, 2024-01-21:
flashed the newest repository on my esp32 and get an bootloop with Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

`Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
*wm:AutoConnect 
*wm:[ERROR] hostname: set failed!
*wm:Connecting to SAVED AP: Vogelgsang WLAN
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.0.199
init plugin: Draw id: 1
init plugin: Breakout id: 2
init plugin: Snake id: 3
init plugin: GameOfLife id: 4
init plugin: Stars id: 5
init plugin: Lines id: 6
init plugin: Circle id: 7
init plugin: Rain id: 8
init plugin: Firework id: 9
init plugin: PongClock id: 10
init plugin: Big Clock id: 11
init plugin: Clock id: 12
init plugin: Weather id: 13
init plugin: Animation id: 14
init plugin: Ticking Clock id: 15
restore plugin: 0
activate: Draw
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400d789c  PS      : 0x00060034  A0      : 0x80081087  A1      : 0x3ffbfef0
A2      : 0x3ffc1bf0  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffb1b70  
A6      : 0x00000000  A7      : 0x00000005  A8      : 0x80081004  A9      : 0x00000001
A10     : 0x3ffc1bf0  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ff42000  A15     : 0x700000bb  SAR     : 0x00000018  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008f3ab  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d789c

ELF file SHA256: 0000000000000000

Backtrace: 0x400d789c:0x3ffbfef0 0x40081084:0x3ffbff10 0x400813ad:0x3ffbff30 0x40087389:0x3ffbff50 0x4008f3a8:0x3ffb1bf0 0x4008fae9:0x3ffb1c10 0x400882e0:0x3ffb1c30 0x4016575d:0x3ffb1ca0 0x4016401d:0x3ffb1cc0 0x40164649:0x3ffb1ce0 0x40163288:0x3ffb1d60 0x40163337:0x3ffb1de0 0x401623fd:0x3ffb1e40 0x40162959:0x3ffb1e80 0x400f0611:0x3ffb1ea0 0x400d7b92:0x3ffb1ed0 0x400d4d51:0x3ffb1f00 0x400d1f01:0x3ffb1f20 0x400d1f27:0x3ffb1f40 0x400d1fc9:0x3ffb1f60 0x400d2f2d:0x3ffb1f80 0x400f3b52:0x3ffb1fb0 0x40089f0e:0x3ffb1fd0
  #0  0x400d789c:0x3ffbfef0 in Screen_::getRotatedRenderBuffer() at c:\users\jan\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ext/new_allocator.h:110
  #1  0x40081084:0x3ffbff10 in Screen_::onScreenTimer() at c:\users\jan\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ext/new_allocator.h:110
  #2  0x400813ad:0x3ffbff30 in __timerISR at C:\Users\Jan\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-timer.c:174
  #3  0x40087389:0x3ffbff50 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x4008f3a8:0x3ffb1bf0 in esp_rom_spiflash_read_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:662
  #5  0x4008fae9:0x3ffb1c10 in esp_rom_spiflash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:662
  #6  0x400882e0:0x3ffb1c30 in spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:599
  #7  0x4016575d:0x3ffb1ca0 in nvs::nvs_flash_read(unsigned int, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_ops.cpp:76
  #8  0x4016401d:0x3ffb1cc0 in nvs::Page::readEntry(unsigned int, nvs::Item&) const at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:871
  #9  0x40164649:0x3ffb1ce0 in nvs::Page::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int, unsigned char, nvs::VerOffset) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:871
  #10 0x40163288:0x3ffb1d60 in nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:579
  #11 0x40163337:0x3ffb1de0 in nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:579
  #12 0x401623fd:0x3ffb1e40 in nvs_get_str_or_blob(unsigned int, nvs::ItemType, char const*, void*, unsigned int*) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:547
  #13 0x40162959:0x3ffb1e80 in nvs_get_blob at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:551
  #14 0x400f0611:0x3ffb1ea0 in Preferences::getBytes(char const*, void*, unsigned int) at C:\Users\Jan\.platformio\packages\framework-arduinoespressif32\libraries\Preferences\src/Preferences.cpp:523
  #15 0x400d7b92:0x3ffb1ed0 in Screen_::loadFromStorage() at c:\users\jan\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ext/new_allocator.h:110
  #16 0x400d4d51:0x3ffb1f00 in DrawPlugin::setup() at .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:82
  #17 0x400d1f01:0x3ffb1f20 in PluginManager::setActivePlugin(char const*) at src/PluginManager.cpp:107
  #18 0x400d1f27:0x3ffb1f40 in PluginManager::setActivePluginById(int) at src/PluginManager.cpp:107
  #19 0x400d1fc9:0x3ffb1f60 in PluginManager::init() at src/PluginManager.cpp:107
  #20 0x400d2f2d:0x3ffb1f80 in setup() at src/main.cpp:187
  #21 0x400f3b52:0x3ffb1fb0 in loopTask(void*) at C:\Users\Jan\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:18
  #22 0x40089f0e:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)`

Janideluxe, 2024-01-21:
If i disable define ENABLE_STORAGE the bootloop doesnt appear

ph1p, 2024-01-29:
I'll close this issue, because you solved it. It seems like your ESP32 does not support the feature of storing data.