ph1p/ikea-led-obegraensad #98: API support

gelse, 2024-01-28:
Some people mentioned it already and a rudimentary API is already implemented with the message and graph support, but i would really like to see a way to control it via any form of API.  

For example a RESTful interface could replace the message endpoint and enable to use it with Home Automation software like HomeAssistant.

Another possibility, tho i see those as two different topics, would be to add MQTT support.

kohlsalem, 2024-01-28:
@gelse which principle of RESTful API did the message API break? I meant it to be used with Home automation; what blocks you there?

For MQTT, idk. While beeing a fan of MQTT, somehow the message usecase does not feel like being a bit benefit to me.

What other control would you imagine? Creating new endponts is pretty simple...

gelse, 2024-01-28:
@kohlsalem well, that was not meant as a "it does break REST", it was meant as "it does not support full control". I really like to be able to switch plugin, set dimmer, etc.

but ... well, if you are asking ... I guess, as "message" is changing the state of the target, it rather should be POST and not GET, but tbh, that's nitpicking and i dont care as long as it works. ;-)

kohlsalem, 2024-01-28:
Hmm. I am failing currently with a proper implementation of a night mode. Moving this beater to homeassistent instead sounds like an interesting idea to me…

and yes, you got me on the post, probably ;-)

can you provide me with a list of all things you want to see?

- Activate Plugin 
- Next Plugin 
- Previous Plugin 
- set Brightness 
- Nightmode on/off (switch off and stay)

kohlsalem, 2024-01-28:
I just noticed- I never checked the communication form the gui with the backend. Shouldn’t there be already something for plugins, brightness, rotation?

gelse, 2024-01-28:
- get plugin list (id and name)
- set plugin by id
- set brightness (parameter: 0-255)
  - 0 can be seen as manual nightmode
- set nightmode duration (parameter: n minutes)
  - would switch to brightness 0 for n minutes
- set orientation (parameter 0-3)

this API is called by a machine, so "convenience" is not much of an issue, therefore i dont see much use of rotate left/right, set orientation makes more sense to me. The same goes for "next/prev plugin" - which is nice for a user, but does not make much sense in an api.
But in the api you definitely need the list of the available plugins.

I will think a bit about more things that would make sense tomorrow.

gelse, 2024-01-28:
> I just noticed- I never checked the communication form the gui with the backend. Shouldn’t there be already something for plugins, brightness, rotation?

i think that's solved via a websocket, which is not really useful for a restful api - i think.
but the endpoints should be internally available already i think

kohlsalem, 2024-01-29:
I created a PR, https://github.com/ph1p/ikea-led-obegraensad/pull/99/
not tested yet, but feel free to.

some thoughts:

- No night mode yet - why would'nt i just set the brightness to 0?
- No orientation yet. I can not imagine why i would ever set the orientation after initial setup. 
- No Plugin Parameters - Plugins could easily register there own endpoints. "AsyncWebServer server;" is  external, so i see no restriction.

gelse, 2024-01-29:
> To set an active plugin by ID, make an HTTP GET request to the following endpoint:
> http://your-server/setplugin

yikes ...
that (and setbrightness) really should be a POST...
a GET never should have any effect on the server.

gelse, 2024-01-29:
ok, all 3 new endpoints seem to work fine.

i just ... HAVE to make a full review.

the response on false values should be valid json, the response code should not be 404 but 422

and getbrightness and getplugin is propably missing, it's not really of much use, but it makes sense to have a getter for a setter.

kohlsalem, 2024-01-29:
NP :-)
422 - Unprocessable Entity
404 - Not Found
I initially lean towards 404 for its logic, but if you prefer 422, that works for me. So does the reasoning for JSON,

While I appreciate the consideration for beauty and symmetry, our ESP32 is running low on memory, and I'd rather not expend it on unnecessary elements.

kohlsalem, 2024-01-29:
> > To set an active plugin by ID, make an HTTP GET request to the following endpoint:
> > http://your-server/setplugin
> 
> yikes ... that (and setbrightness) really should not be a GET... a GET never should have any effect on the server.
> 
> i would have used a POST, but as i learned some weeks ago, the better verb is even PATCH.

yes, im fully aware of post. But then i (and everybody) have to se wget instead of a browser to test. 

Thats why i have it on get. The change would be obviously trivial, but...

gelse, 2024-01-30:
But wget actually supports POST? (And curl would be better anyway - also there are really easy tools like insomnia, or my favourite: Bruno)

gelse, 2024-01-30:
> NP :-)
> 422 - Unprocessable Entity
> 404 - Not Found
> I initially lean towards 404 for its logic, but if you prefer 422, that works for me. So does the reasoning for JSON,
>
> While I appreciate the [...]

422 seems to be the broadly accepted value for such things. Anyway, 404 is just wrong, it's purpose is to indicate that the endpoint is not there, not that parameters are missing or wrong.

But I see your point about the low memory and agree with you there.

kohlsalem, 2024-01-30:
All possible. Thats how its implemented in the websocket implementaion, with a "big" json.

Still i would somehow prefer something like "Get Status", returning brightness and active plugin...

kohlsalem, 2024-01-30:
so, getstatus instead of "get plugin list"

gelse, 2024-01-31:
oh ok, i did not see you already started work on the code.

anyway, i took the challenge and implemented it myself, the PR is here: #100 
for my tastes, my code improved yours a bit:
- removed data from getStatus and made a separate endpoint which makes usage of the AsyncResponseStream.
- cleaned up the response messages (no need to send "OK" if the status is 200)
- added a tiny bit of error handling and prettier error messages
- moved the implementation of the endpoints to the messages file

ph1p, 2025-01-09:
I will close this for now, reopen it again if you want :)

closed 2025-01-09 by ph1p