ph1p/ikea-led-obegraensad #101: Active pixels flash once at max brightness every minute

scfede, 2024-02-02:
I noticed that whenever the minute changes, all active pixels flash once at max brightness and then dim again to stay on the desired/set brightness. Saw that with "Big Clock" but also with other plugins, so i'd guess this is a global issue.

How to reproduce: Use the plugin "Big Clock", set brightness to a very low value and wait until the minute flips to the next one.

dahabakuk, 2024-09-27:
i noticed this too. still looking around in the source to get a better understanding why this happens.... its really annoying :)

BrookeDot, 2025-06-23:
I recently experienced this too, I think it could be that the brightness is being set to 255 when `DrawBigNumbers` is called:
https://github.com/ph1p/ikea-led-obegraensad/blob/main/include/screen.h#L75

BrookeDot, 2025-06-24:
Thanks for testing @TheMysticle, the brightness change must be further up in the stack then :-/

TheMysticle, 2025-06-25:
All good, I'll keep trying to get it working but I don't know if I'll be able to. This isn't my strong point 😂 It seems I'm able to eliminate the flashing by using entirely software PWM, however then a new issue arises because the entire thing flickers subtly and randomly at lower brightness. It's subtle but it's definitely there and constant.

TheMysticle, 2025-06-27:
> Thanks for testing [@TheMysticle](https://github.com/TheMysticle), the brightness change must be further up in the stack then :-/

Okay, good news. I have found the issue. It's actually a lot easier than we may have thought. It's the loop that looks for messages once a minute. That's what causes it. So just remove the line:

`Messages.scrollMessageEveryMinute();`

From main.cpp

If you need to actually project messages onto the display a much better way would be to make a plugin specifically for that purpose :)