ph1p/ikea-led-obegraensad #108: Reconnect to WiFi in case of disconnection

c64emulator, 2024-04-13:
Es gab bereits schon mal ein Ticket und einen PR dazu. Das Problem scheint nicht behoben bzw. wieder da zu sein:
Leider musste ich bei meinem neuen WLAN-Router einen Scheduler einrichten, der in jeder Nacht (nachdem die FRITZ!Box einen DSL-Reconnect durchgeführt hat) den Router neu bootet. Das war notwendig, weil das WLAN extrem unzuverlässig lief. 
Als Folge davon öffnet das LED-Panel nun seinen AP (mit Webserver auf 192.168.4.1). Allerdings zeigt es dort dann die normale GUI (und nicht die WiFi-Konfiguration).

_Originally posted by @c64emulator in https://github.com/ph1p/ikea-led-obegraensad/issues/14#issuecomment-2052633457_

c64emulator, 2024-04-21:
small modification in "connectToWiFi()" fixed it for me:

>   wifiManager.setConfigPortalTimeout(300);
  bool res;
  res = wifiManager.autoConnect(WIFI_MANAGER_SSID, WIFI_MANAGER_PW);
  if (!res)
  {
     Serial.println("***** Failed to connect - RESTARTING...");
     ESP.restart();
  }

WRedux, 2024-06-06:
Hallo,

bin C-technisch nicht so gut drauf....

Kannst mir bitte sagen, wo und wie Du die "connectToWiFi()" modifiziert hast?
Einfach dazuhängen ging nicht...

Danke!

c64emulator, 2025-02-20:
Ich habe den aktuellen Code nochmal komplett ausgecheckt und neu gebaut. Für mich ist das Problem behoben. Ich würde nur empfehlen, ein "WIFI_MANAGER_PW" zu nutzen.

closed 2025-02-20 by c64emulator