ph1p/ikea-led-obegraensad #109: Display flickers when using PWM

Axelmannen, 2024-04-14:
Everything works as it should in all plugins that only use full brightness, but for every dimmed pixel added more "flickering" noise is introduced. This noise consists of random low brightness pixels and can be all over the screen.

I've had it working before and I have no clue what is causing it. Has anyone else had the same problem?

Axelmannen, 2024-04-14:
It seems to be an issue with my local hardware

closed 2024-04-14 by Axelmannen

cubic3d, 2025-03-06:
@Axelmannen Can I ask what the issue was in your case? I have tried with two of my LED Panels and two completely different boards (Lolin S3 Mini and NodeMCU32 style one with shielded module). Both constellations show flickering as you described if reducing brightness. If I plug the cable through the boards USB, it seems fine. Also reducing the PWM frequency to 100Hz fixed it almost (didn't with higher) but noticeable on already dimmed pixels.

Axelmannen, 2025-03-07:
I think what caused me to open this issue was something different, with random noise showing up all over the screen, even on pixels that were supposed to be black. 

But yeah in my experience you can't really use global dimming at the same time as local dimming, if that makes sense. That causes the low-brighness pixels to flicker. But I've been abroad for some time now so don't remember exactly and things might have changed since.