ph1p/ikea-led-obegraensad #110: Another Noob needs some help...

WRedux, 2024-04-15:
Hello there,

guess, I'm more noob than the others.... 
I can compile, but the Code wont upload to the ESP.

Shortly, after the Message "Connecting..." displays, it stops showing:

A fatal error occurred: Packet content transfer stopped (received 8 bytes)
*** [upload] Error 2
============================================== [FAILED] Took 19.63 seconds ==============================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:19.629
========================================= 1 failed, 0 succeeded in 00:00:19.629 =========================================

 *  The terminal process "C:\Users\x\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--upload-port', 'COM7'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

I've chosen the right COM-Port.
I did not changed the Board, assuming I use the right one (ESP WROOM 32)

I guess, I miss some Basics. I'm not new to Arduino and ESPs, I used the both with ArduinoGUI.

[image]

c64emulator, 2024-04-21:
https://github.com/espressif/esptool/issues/741

BrookeDot, 2024-12-04:
@WRedux do you mind marking this as closed as it sounds like you have a working solution? Thanks!

closed 2025-02-07 by WRedux