ph1p/ikea-led-obegraensad #111: Problem that all LEDs are glimpsing

hobbid, 2024-04-21:
Hi, 
i built your project and after some pitfalls i got it up and runing but i have an brightnes issue. it seems that all leds glimms and therefore there is no clear contrast an the quality of the clock or other screens are really bad. switching the brightness in the web frontend down not really help. 

To get it running i switched to pins as PIN 16 for the button was not available on my esp32 and with PIN 12  i had issues booting. After googlind i found out that for Pin12 it is stated: "boot fails if pulled high, strapping pin" so i switched to PIN 15.
so this is my pinout now.
`#ifdef ESP32
#define PIN_ENABLE 26
#define PIN_DATA 27
#define PIN_CLOCK 14
#define PIN_LATCH 25
#define PIN_BUTTON 5
#endif`

This is the board i used 
[image]

And here is a image of the glimpsing leds:
[image]

Any idea what is wrong why all leds are glimpsing ? 

Best regards

Dirk

jaal2001, 2024-04-21:
Did you unsolder the chip on the display completely or just one pin? There have been cases, where the chip on the display needed to be removed completely.

hobbid, 2024-04-21:
I desoldered the whole chip but unfortunately I ripped the first pad. Is that an issue ?

[image]

jaal2001, 2024-04-21:
Probably not, but if you look right of your cable, it seems that there is a wire shortig multiple soldering pads. Maybe that is the issue.

hobbid, 2024-04-22:
That did the trick ! With the pure eyes the wire was not really noticeable. Posting the photo and the comment from @jaal2001 did the trick! Thanks ! 🙏 
[image]

closed 2024-04-22 by hobbid