ph1p/ikea-led-obegraensad #113: ESP32-C6 support with Arduino-ESP 3.0.0?

derbirch, 2024-04-27:
I'm totally new to ESP dev and PlatformIO. With basic knowledge in C, I tried to twist some fun to meet the new Arduino 3 & ESP IDF 5.1 API changes, but the SDK still seems to be a big issue when I try to build.

Is there any suggested `platform.ini` config for the `esp32-c6-devkitc-1` board? Or do I have to pick other esp32 boards? If so, do you have any recommendations? I struggle to pick esp32 boards as there are too many SoC models/variations and different boards with them.

Thx in advance.

BrookeDot, 2024-12-04:
Hi @derbirch I recently built another PlatformIO project (ESPHome) on the ESP32-C6 and I'll admit it was not very straight forward, if you are new to the ESP I may recommend a board that has been around for a while[. The Readme has a few recommendations](https://github.com/ph1p/ikea-led-obegraensad?tab=readme-ov-file#pins), I just updated my NodeMCU and noticed that the 4MB is getting filled, so you may want to go with an 8MB board if you have the budget.

Beyond that, I would recommend sticking with the ESP32 over the 8266 due to the increased RAM, you'll just have a better time :-) So that would leave you with the Lolin D32 (Pro), which I haven't used, but looks like a nice board.

If it's helpful, here is the ESPHome config (they use YAML) that did end up compiling:
```
esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
```

derbirch, 2024-12-04:
> Hi @derbirch I recently built another PlatformIO project (ESPHome) on the ESP32-C6 and I'll admit it was not very straight forward, if you are new to the ESP I may recommend a board that has been [...]

Thx for your reply.

I have bought another ESP32 board.

With some twist in PIN def, the firmware works well except memorizing my drawings.

BrookeDot, 2024-12-05:
Do you mind marking this issue as solved for now if things are working on your ESP32 board?

derbirch, 2024-12-05:
> Do you mind marking this issue as solved for now if things are working on your ESP32 board?

It's an original ESP32, not C6, sorry if I make any confusion. But I will try to rebuild it for C6 with updated dependancies when available.

ph1p, 2025-01-09:
Simply open it again when it is needed

closed 2025-01-09 by ph1p