ph1p/ikea-led-obegraensad #116: Plattform IO Question an Binary HEX Upload how does it work

raindancer2204, 2024-08-26:
Hello, really a great project here and a great job.

Now I have to ask a “beginner’s question” about PlatformIO and compiling. So that I can get the HEX or binary file for flashing described ESP 32, do I compile it in the IO platform?  Or am I misunderstanding that?

If the fcompiled hex file (binary) compilation is done how will it transfared to the the ESP via platform IO and a Plugin?

Isn't it also possible to provide a HEX file with the current build and default settings for the ESPs described?

The instructions are really great so far, maybe you could add 2 more sentences.

Thank you

Your Raini

JulianBastel, 2024-09-28:
Thank you for that information. I flashed with esptool.py but I was not working. Via PlatformIO it is working.

BrookeDot, 2024-12-04:
While I agree the documentation could be improved, one of the tricky things is not duplicating and attempting to maintain information that has a source of truth. 

In this case, what are your thoughts on linking to the official PlatformIO documentation for many of these steps? For example, I used the following quite heavily when getting started:
https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode

Another consideration is platform specific documentation, such as COM port for Windows vs dev/ttyUSB0 for MacOS/Linux. 

Again I agree with you, but would like some further discussion before taking a stab at the readme updates.

ph1p, 2025-01-11:
Updated the readme. If you want to update it by yourself, just open a PR :)