ph1p/ikea-led-obegraensad #121: Compilation errors and boot looping on Wemos D1 mini (clone)

ilendemli, 2024-10-22:
Hi!

Besides after fixing the errors in main.cpp#4 (typo in ifdef) and adding ESPAsyncUDP for the ESP8266 and compiling, my board gets stuck in boot loop and I have no idea why. Tried on two different 8266 development boards and with various changes in configuration. Compiled and uploaded both on Windows and MacOS, no change.

```bash
 *  Executing task in folder ikea-led-obegraensad-feature-vscode_devcontainer: platformio device monitor --environment d1_mini --port /dev/cu.usbserial-1420 

--- Terminal on /dev/cu.usbserial-1420 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000b7020
~ld
���g�;��o|�
           $�$`c�
                 |s�l�'��'�l`��;�l�d
                                    �
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000b7020
~ld
����n�;��n|�
            $�d`c�
                  <r�l�'��'�d`��s�$�l
                                     �
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000b7020
~ld
���o�;��n|�
           $�d`c�
                 <;�l�g��g�l`��;�l�d�
                                     �
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000b7020
~ld
���'��{��'<�$
             �l`c�
                  |s�l�'�
                         �n�
                            $ ��{�$�l�
                                      �
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000b7020
~ld
����n��{��n|�
             $�d`c�
                   |s�d�n�
                          �n�
                             $ ��s�$�l
                                      �
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000b7020
~ld
���o�;��o|�
           $�d`c�
                 |s�d�n�
                        �n�
                           $ ��s�$�l

                                    �
```

BrookeDot, 2024-12-05:
If you are still having issues maybe post a `diff` of what you have changed (removing any passwords), as that could help others replicate the results.

kd00r, 2025-01-11:
I have the same error as OP. I also tried with two different ESP8266 (One NodeMCU 0.9 and one clone of the former).

coelner, 2025-03-16:
The watchdog gets triggered: https://www.espressif.com/en/support/documents/technical-documents?keys=&field_type_tid%5B%5D=14&field_download_document_type_tid%5B%5D=516