ph1p/ikea-led-obegraensad #126: Stars animation brightness is not smooth anymore

jekkos, 2025-01-11:
Today I built latest master version and upgraded the ESP32. Now I see the stars animation is not smooth anymore, seems that it's not as smooth as it was before. Maybe I did this a bit too soon as I see now more updates came in.

ph1p, 2025-01-12:
Thanks @jekkos :)
Hopefully the latest commit fixed it for you. Could you please test it?

JanPetterMG, 2025-01-14:
Backported all new commits since new year, and I'm seeing the same issues.
Best seen on standstill pixels, even at max global brightness, every grayscale pixel is flickering like crazy. The PongClock is one of the most affected plugins I'm aware of.

Had an quick troubleshooting session, but I couldn't pinpoint the root cause, as there's too many commits recently. Isolated and eliminated the new transformation/rotation method to no avail.

The best clue I've got so far is a bit awkward, something has affected the `renderBuffer_` array itself.
The first method works as it always has, but the latter one (custom) stopped working as soon as I backported. Everything else works, although the performance has gotten rubbish.
````
void Sreen_::clear()
{
  memset(renderBuffer_, 0, COLS * ROWS); // No issue
}

void Sreen_::fill()
{
  memset(renderBuffer_, 255, COLS * ROWS); // Crashes and hangs
}
````

*Disclaimer: I'm running a fork.*

closed 2025-01-15 by ph1p

jekkos, 2025-01-18:
Indeed seems to be resolved with the latest build. Thanks for this!