ph1p/ikea-led-obegraensad #141: Cannot find Web ui

Saber4952, 2025-04-03:
How do I find the ip to access the web ui? 
The instructions say "You can get the IP via serial output or you can search it in your router settings." But neither of those work for me.
Is there a location in the code that says the ip?

folkertvanheusden, 2025-04-07:
Try looking in the dhcp table of your router.

Saber4952, 2025-04-07:
I do not have access to that so i am looking for an alternative way to find it

folkertvanheusden, 2025-04-07:
What did you put in `#define WIFI_HOSTNAME ""` in `secrets.h`?
Try pinging that. If that doesn't work, add `.local` to the hostname you're pinging and try again.

Saber4952, 2025-04-07:
There is nothing in that was i supposed to put something there?
also i am able to get into the setup wifi to connect it to a network and it works but im looking for the web gui to control it fyi

closed 2025-04-07 by Saber4952