ph1p/ikea-led-obegraensad #142: Message doesn't disable Pong Clock

PiotrMachowski, 2025-04-05:
When Pong Clock is running and `/api/message` endpoint is used both of them are displayed at the same time - only clock part disappears, pong part is still running.

BrookeDot, 2025-06-08:
This is the way the message plugin is supposed to work (for better or for worse). 

Could you describe what you were hoping the message plugin would do? It sounds like you'd like it to switch to a new plugin mode?

PiotrMachowski, 2025-06-08:
Well, at this moment message is displayed over the dynamic content so everything ends up as a flickery mess. I solved it in my fork by creating "Empty Screen" plugin and turning it on for a duration of a message

BrookeDot, 2025-06-08:
> at this moment message is displayed over the dynamic content so everything ends up as a flickery mess.

Yeah, that's why I don't use the message plugin, so glad you were able to solve it. I think @kohlsalem and @gelse  intended the API to be simple (see #100) but as you mentioned, it doesn't always work with dynamic content.

>  I solved it in my fork by creating "Empty Screen" plugin and turning it on for a duration of a message

Would a simple PR into this project be possible, maybe it could be an option in the config to enable your plugin?

PiotrMachowski, 2025-06-08:
My fork contains several other functionalities that make it more useful for Home Assistant users, I plan to make add them to this project, but I would have to split it into several PRs. The other part is that I run out of memory after including all my changes, so I had to disable plugins that are not useful for my use case. 

At this moment you can check it out here:

https://github.com/PiotrMachowski/ikea-led-obegraensad